shaunlowis/rocketeering #13: Perform mods to airframe

AngusEason, 2024-10-12:
- Load calcs for avbay mounting
- Drill holes for mounting avbay
- Print updated avbay and install M4 heat set inserts
- Test fit mounting avbay
- Install thermocouple wire and connectors down to the engine (dilling as required).
- Secure wires with epoxy

shaunlowis, 2024-11-26:
- Test thermocouple connector mount, see [here](https://cad.onshape.com/documents/872890136eee2b11197cc0fd/w/8573091cfe1338d2f6237b9b/e/51cf60da6aae13c0dbae036a)
- Made new drill jig, fit to air frame and test, see [here](https://cad.onshape.com/documents/872890136eee2b11197cc0fd/w/8573091cfe1338d2f6237b9b/e/d5691ce231b18a61533ffa2f)

shaunlowis, 2024-12-18:
- Holes drilled by big A
- Dremelled a window for the camera, looks good
- Thermocouple TODO

shaunlowis, 2025-01-22:
Donezo, all mounted.

[image]

closed 2025-01-22 by shaunlowis